SUPLA/supla-device #5: can't see value from SUPLA_CHANNELTYPE_GENERAL_PURPOSE_MEASUREMENT

Wojaran, 2022-08-15:
I created

Supla::Sensor::AirQualitySensor::AirQualitySensor() : lastReadTime(0) {
  channel.setType(SUPLA_CHANNELTYPE_GENERAL_PURPOSE_MEASUREMENT);
  channel.setDefault(SUPLA_CHANNELFNC_GENERAL_PURPOSE_MEASUREMENT);
}

i registered and stuff i see it on list in supla cloud but i dont see values... anything I need to do?

Channel(0) value changed to 1204.0
Sending: [53 55 50 4C 41 10 4E 0 0 0 67 0 0 0 E 0 0 0 0 0 2C 1 0 0 0 0 0 0 0 D0 92 40 53 55 50 4C 41 ]
Sending: [53 55 50 4C 41 10 4F 0 0 0 28 0 0 0 10 0 0 0 FF 1 0 0 0 0 0 0 3B 5B 5 0 0 0 0 0 53 55 50 4C 41 ]
Received: [53 55 50 4C 41 10 10 0 0 0 32 0 0 0 10 0 0 0 5F 7C FA 62 0 0 0 0 A4 57 6 0 0 0 0 0 53 55 50 4C 41 ]
Channel(0) value changed to 425.0
Sending: [53 55 50 4C 41 10 50 0 0 0 67 0 0 0 E 0 0 0 0 0 2C 1 0 0 0 0 0 0 0 90 7A 40 53 55 50 4C 41 ]

klew, 2022-08-15:
Hi, general purpose measurement channel is not supported yet by Supla server.
Currently people use thermometer channel type for this purpose as a workaround

Wojaran, 2022-08-15:
Hmm ok, I thought if I see panel for it it will work:) any idea if it will be added/plans? Thanks for info

klew, 2022-08-16:
I hope that this will be The Next Big Feature in one of the nearest server/cloud release. Content for next two months on work is more or less already planned, so my expectation is to get it somewhere in early 2023. However this is only my guess.

closed 2022-08-16 by klew