SUPLA/supla-device #17: Hygrometer - sent values not visible in Supla cloud

milanos, 2022-09-04:
I create new class based on HygroMeter (hygro_meter.h)class.
I see in log sent values 
[image]

but In Supla I can see any measurements. I see still "-1"
[image]

if I use the thermometer sensor (thermometer.h), the data is displayed correctly but but shown as a temperature sensor

milanos, 2022-09-04:
never mind, I used examples from it: https://forum.supla.org/viewtopic.php?t=8366 and it's work.

milanos, 2022-09-05:
I reopen, because founded solution use function with 2 variables: 
channel.setNewValue(HUMIDITY_NOT_AVAILABLE,getHumi()); and it works but not entirely correct. It solution send tem and humidity for the humidity sensor only.

I'd like use function:  channel.setNewValue(getHumi()); but in SUpla clud I see indication= -1, on smartphone I see inicator 1271310,3 :(

klew, 2022-09-05:
Humidity only channel uses the same data structure on Supla protocol as humidity+temperature sensor. However server ignores temperature setting and present only humidity value.
That's why you have to use: channel.setNewValue(-273, getHumi())

Missing chart for humidity sensor is a known issue.
Currently it should be available on Cloud interface (https://github.com/SUPLA/supla-cloud/issues/546)
In case of mobile apps, this is still not implemented yet. Here is issue to track it: https://github.com/SUPLA/supla-android/issues/307

milanos, 2022-09-05:
OK, thank you.
Yes, the effects are the same when using
channel.setNewValue(-273, getHumi())
or 
channel.setNewValue(HUMIDITY_NOT_AVAILABLE,getHumi()) 
HUMIDITY_NOT_AVAILABLE is constans value set in code.

it remains to wait for the solution of the graph visibility problem

closed 2022-09-05 by milanos